exploitagency/ESPloitV2 #2: Setting delay values anything below 1020ms doesn't seem to work.

jamesmcintyre, 2017-08-10:
Whether in the configuration page or in payloads or live payload, setting "DefaultDelay" or "CustomDelay" to anything below 1020 does not work- as in it seems to cause the execution of the lines to fail and even 1020 doesn't work all the time.

I found this line, is there some inherent reason this device can't handle fast line entry? Seems like it shouldn't be a problem but I'm no arduino-master, far from it!

https://github.com/exploitagency/ESPloitV2/blob/b744e22c0fd963f887dcc51aeb6638a550176fcb/source/ESP_Code/ESP_Code.ino#L127

exploitagency, 2017-08-17:
I plan on improving the speed in the future but you should focus on making small concise payloads for now.

Check this example out

Rem: Apple's OS X does not allow plugging in a keyboard and immediately pressing buttons.
Rem: It requires Right of Left Shift Key, "z", and Left of Right Shift Key, "/", to be pressed to identify the keyboard layout as English.
Rem: This payload is a workaround and is only needs to be ran the first time the device is inserted, on future inserts this is not required.
Rem: Instructions.
Rem: Set this payload to run as automatic and insert the device or run it through the browser.
Rem: Manually click "Continue", the payload will press the required keys to identify the keyboard.
Rem: Now manually click "Done".
DefaultDelay:2000
CustomDelay:5000
Press:122
CustomDelay:3000
Press:47

The Rem and default delay are not needed(set delay in config page)  default delay in a payload is only an over ride so remove them as they also cause a tiny delay in your payload script for wvery rem/line you add.  That will speed the payload up.  But check for updates every so often. Things will be improved over time.

jamesmcintyre, 2017-08-31:
Thanks @exploitagency. My main goal is to be able to run 10-100 lines which are an alphanumeric group of characters each line with carriage return at the end and to be able to do this fast (as fast as the software/ui being controlled will consistently allow). As it is now when I write a .txt file to do this it runs pretty slow, line-by-line. Is there a way I can speed this up now or do improvements need to be made first?

exploitagency, 2017-09-04:
I released this part early just for you...
https://github.com/exploitagency/ESPloitV2/blob/master/source/Arduino_32u4_Code/Arduino_32u4_Code.ino

Use 75ms delay and I think you will be safe

exploitagency, 2017-09-08:
I think this can be closed now.  Let me know how it goes.

closed 2017-09-08 by exploitagency

jamesmcintyre, 2017-09-12:
@exploitagency Just saw this, thanks Corey this is awesome! I'll try this as soon as I can!

exploitagency, 2017-09-12:
No problem, let me know how it goes.

exploitagency, 2017-09-27:
Check out the latest release for both the 32u4 sketch and the ESP sketch.  There have been a lot of improvements since I closed this issue.  Speed improvements, WiFi based exfiltration options(HTTP,FTP),Captive Portal Credential Harvester(Phishing), bug fixes, etc...

Also I have now worked up a ducky script to ESPloit converter: https://github.com/exploitagency/Duckuino

Just wanted to make sure your payload is a success.

jamesmcintyre, 2017-09-28:
@exploitagency awesome! I will try this hopefully by next weekend! Awesome work! And I'm sure the ducky to ESPloit converter will be very valuable!