exploitagency/ESPloitV2 #3: [enhancement] Changeable USB VID & PID

whid-injector, 2017-08-10:
Implement the Changeable USB VID / PID feature in order to let user decide its favorite VID/PID, in order to bypass endpoint restrictions.

exploitagency, 2017-08-10:
This is easy enough to do by the end user.

Add this to the end of boards.txt and select Cactus WHID under boards instead of LilyPadUSB when you upload the Arduino_32u4_Code sketch.

`##############################################################

CactusWHID.name=Cactus WHID
CactusWHID.vid.0=0x1B4F
CactusWHID.pid.0=0x9207
CactusWHID.vid.1=0x1B4F
CactusWHID.pid.1=0x9208

CactusWHID.upload.tool=avrdude
CactusWHID.upload.protocol=avr109
CactusWHID.upload.maximum_size=28672
CactusWHID.upload.maximum_data_size=2560
CactusWHID.upload.speed=57600
CactusWHID.upload.disable_flushing=true
CactusWHID.upload.use_1200bps_touch=true
CactusWHID.upload.wait_for_upload_port=true

CactusWHID.bootloader.tool=avrdude
CactusWHID.bootloader.low_fuses=0xff
CactusWHID.bootloader.high_fuses=0xd8
CactusWHID.bootloader.extended_fuses=0xce
CactusWHID.bootloader.file=caterina-LilyPadUSB/Caterina-LilyPadUSB.hex
CactusWHID.bootloader.unlock_bits=0x3F
CactusWHID.bootloader.lock_bits=0x2F

CactusWHID.build.mcu=atmega32u4
CactusWHID.build.f_cpu=8000000L
CactusWHID.build.vid=0x0000
CactusWHID.build.pid=0xFFFF
CactusWHID.build.usb_product="Cactus WHID"
CactusWHID.build.usb_manufacturer="April Brother"
CactusWHID.build.board=AVR_LILYPAD_USB
CactusWHID.build.core=arduino
CactusWHID.build.variant=leonardo
CactusWHID.build.extra_flags={build.usb_flags}`

[image]

[image]

[image]

[image]

Replace this portion with your spoofed VID/PID.
`CactusWHID.build.vid=0x0000
CactusWHID.build.pid=0xFFFF
CactusWHID.build.usb_product="Cactus WHID"
CactusWHID.build.usb_manufacturer="April Brother"`

whid-injector, 2017-08-10:
Cool solution!
I never had time (or need yet) to dig into this, and didn't want it to get lost in the issues. (reason why I opened here, since ESPloit V2 will be delivered with Cactus WHID by default).
Glad I opened the issue and you immediately proposed a solution! Nice job man!

exploitagency, 2017-08-10:
Speaking of which ask April Brother to ship with the latest firmware as I set it up to connect to my server when you go to Update Firmware and check for updated releases on Github, display if an update is available, then download the update and apply it if the user chooses to.

This will make it simple for an end user to keep in the loop without needing any knowledge about Arduino or Github etc... all the old features are retained, this is just a bonus.

whid-injector, 2017-08-10:
This feature is very cool and I like it! However, might be better to be configurable on the UI (i.e. enable/disable autoupdate check)
Imagine the scenario where a Cactus WHID is deployed during an engagement and is using the client wifi network as Client and the attacker is controlling it within the same LAN... With your auto-update feature on... it would start to generate outbound traffic to the internet (i.e. your server) that would be logged and likely traverse target's IPS/IDS/FW. 
Ideally, is always better to keep these devices as silent and stealthy as possible from a network activity perspective.

exploitagency, 2017-08-10:
It will only be triggered if someone goes to the config page and clicks on update firmware.

whid-injector, 2017-08-10:
Now I got it! So is triggered only under user's action. Cool then!
Yeah, is definitely the best solution indeed!
Well actually the dealing-with-Arduino stuff... was mainly to not keep things that simple, that everyone will be able to use it immediately (the so called plug-n-hack) =P

whid-injector, 2017-08-14:
Forgot to close the ticket :)

exploitagency, 2017-08-15:
I will actually leave this one open so people can find it easier until I get around to adding it to the readme along with changing keyboard layouts.

closed 2017-09-06 by exploitagency