exploitagency/ESPloitV2 #5: [enhancement] Call back C&C

whid-injector, 2017-08-10:
**Just got an idea and throwing here for future development.**
**Scenario:**
- Cactus WHID deployed in an unknown environment (e.g. imagine a weaponized USB gadget delivered by courier)
- Once connected to the USB (i.e. powered ON) the ESP (setup to act in client-mode) starts to scan for Open WiFi APs.
- If found, tries to connect and starts contacting back the C&C over HTTPS on 443 (eventually we can try to use a DNS tunnel over TCP 53 [if ESP supports it])
- If it managed to connect to an Open AP and gets through eventual FW/IPS... voila'! The attacker can control it from its on C&C god-knows-where ;]

P.S. Actually, to be fair, while designing the Cactus WHID I have thought of its upgrade... a Cactus WHID 2G ( by using an Atmega 32u4 + SIM800H 2G module ) but that would kill its small price... :)

exploitagency, 2017-08-10:
You actually can get the 2G modules cheap from certain sources but they are larger in size and finding a 2G network provider may be a chore in some locations.  I wrote ESPloit and designed the Proof of Concept in a single night while watching TV so the code is still based on a really simple project.  ESPloitv2 just refines into something more suitable for production.  I have thought of a C&C client/server deal, maybe later down the road...

whid-injector, 2017-08-10:
Yeah, dimensions and the fact that 2G is going to slowly die worldwide (i.e. in USA I guess is almost gone) refrained my desire of creating this kind of HW. 
I would rather think to add a uSD slot to the Cactus WHID. But the difficult part then, would be to find a suitable USB case at cheap price and a manufacturer that would not ask 1000 MOQ =]

exploitagency, 2017-08-10:
I always thought the 3M spiffs on an ESP-12 was enough to eliminate the need for an SD card for most projects and one reason this specific project was so unique.  But more is always better.  You could have likely have nearly "every payload ever" on an SD Card.

minkione, 2017-08-10:
The uSD would be to be use as diversion in case you use the Cactus WHID as dead-drop USB.
Then the victim will plug in the USB and see a mounted NTFS/FAT FS. Rather than wait for anything to appear. 
Yeah the SPIFFS has plenty of space and the cost itself is really good.
That's why while designing it I preferred to keep the uSD out of scope at least for the version 1.0 :)

ChandraOrbit, 2017-09-28:
I am waiting for a GSM support upgrade, the price issue is not a problem because at least there is a modernization to follow the needs of the current era of mobile gadged everything.

ChandraOrbit, 2017-10-01:
there is an information on [Ryan Ackroyd's tweeter](https://twitter.com/APT1337/status/884782754818396161) with his eye-catching video demonstrating here, because he has already implemented it, it's just that the workings, schemes and source codes are not mentioned. Maybe in time to come you can make it happen. thanks.

whid-injector, 2017-10-01:
I was already making prototypes with such kind of GSM modules, even before the inception of WHID.
The main blocker, is that the board would then not fit a USB case.
At this point, just use a OrangePi IoT 2G with P4wnP1.

whid-injector, 2017-10-01:
@mame82 will be happy if u will port his P4wnP1 to OrangePi IoT 2G 😉

ChandraOrbit, 2017-10-08:
@whid-injector, I happen to have one unused OrangePi IoT 2G.

I would be very grateful to you and how happy I am if you are willing to provide a guide guide or a little hint to install OrangePi IoT 2G with P4wnP1.
I've searched on Google but did not find the clue, and a few days ago I've asked mame82 HERE but the answer is "No port for OrangePi atm"

thanks.

whid-injector, 2017-10-08:
1) Re-read my intial message about it. I suggested you to port it to OrangePi, NOT to ask someone else! 😒
2) Your are polluting this thread with out of topic crap! 😤
3) Here we are not a help desk.
4)Did you first tried to install Armbian on Orangepi IoT and tried to install P4wnP1 and see where is failing?😤

People has a job, family and hobbies. I am not here to do stuff for you!

I am really pissed when I see these kind of messages.😤😤😤

exploitagency, 2017-10-08:
I will look into it if someone wants to donate the hardware and 2G service that works in the US.  But why not try to create something original with broader network support?  2G options in the US are being phased out.  I would prefer to create something original.  It is a low priority on my list of projects but maybe I will get around to it.

whid-injector, 2017-10-08:
From my side, I am going to port P4wnP1 straight to my POTAEbox board once I will bring it to production.
However OrangePi IoT 2G is already a nice bord to play with P4wnP1.
As for MobNets, only US killed 2G IIRC.
Countries like India or China will take other 10 years (at least) before shutting it down. Reason why OrangePi used a 2G modem. (cheap and still used in most of countries)

exploitagency, 2017-10-08:
Well I wish you luck.  I have a lot of other small unrelated projects I am working on though.  So me porting something is not on my to do list, but I have built a simple cellular proof of concept before just never bothered to get a 2G sim to test it.  I have never looked into that Pi project though, theres a ton of attack vectors when you bring a Pi into the equation, I just looked up the P4wnP1 project and there are even more attack vectors than they utilize.  The Pi Zero W is pretty amazing hardware though for its size, but I rarely pull it out.  But my PoC used much simpler 8 bit hardware but that is a neat project, I think your port should be fairly straight forward.  I will check it out when your done.

whid-injector, 2017-10-08:
P4wnp1 is indeed a nice framework. Unfortunately both RPi and OrangePi are too bulky.
That's why WHID + ESPloit is still the best covert solution for remote HID attacks, that can fit an innocent USB case.
P.s. Corey, as soon as I have few prototypes of POTAEbox I am going to send you one, sure you will enjoy it for your cool projects 🙂

https://www.dropbox.com/s/2v6y6wvd1wb5fck/File%2008-10-2017%2C%2012%2022%2013.png?dl=0

exploitagency, 2017-10-08:
Thank you, I am sure I will find it useful.

ChandraOrbit, 2017-10-08:
@whid-injector.  Sorry if what I say does not please you. Maybe I was wrong in shipping. I only connect from the previous message. So I think there is little clue to me. Suppose I was a blind man, if there was a little lightening it would be wise.

Thanks and sorry before and last.

exploitagency, 2017-10-08:
You will do fine  @ChandraOrbit his point was we are not here to spoon feed anyone.  But while we are on the topic of spoon feeding I have tried my best to NoobProof most of my projects so there is hope of replicating them or it would defeat the purpose of sharing them in the first place.

And while we are so far off topic @whid-injector have you tried this out yet?

https://exploitagency.github.io/Duckuino/index.html

It should help many users who are the copy paste type to convert ducky scripts into something to run on the stock firmware.  I don't think many people are aware it exists yet.

whid-injector, 2017-10-09:
Sadly I haven't time yet to test that feature. But is indeed a cool thing to have for newbies! 🤘🙂
Actually as unsatisfied Hak5 customer, I prefer to have nothing to do with them. Even compatibility with their Ducky scripts. 😁

exploitagency, 2017-10-09:
Well everyone is entitled to their own opinion.  I like the Hak5 crew just fine, I have even had projects featured on their web show before.  I actually chose to create my own payload scripting language out of respect to them and their product so I would have something more original.  Plus ESPloit's scripting language is in my opinion a lot more extensive, you can type out strings that include carriage returns, mouse emulation, etc... I figured referencing ASCII codes for button presses would be trivial for the end user but Ducky Script uses plain text for these instead, it is a toss up there.  Though apparently an option for Ducky Script conversion was needed to support the conversion of the massive amounts of existing payloads available for the Rubber Ducky.  Some users are actually migrating our way that have already wrote payloads for the Rubber Ducky.  And some users just want to search for and copy/paste a payload and it work.  Even if we don't use certain tools, some things just have to exist for those specific scenarios so our project can be a success.

mame82, 2017-10-09:
@exploitagency 

I've done an converter for DigiSpark (AVR based), too
https://github.com/mame82/duck2spark/

I even done a reimplementetion of duckencoder:
https://github.com/mame82/duckencoder.py

The former uncovered how bad the original encoder was written.

Finally I incorporated it into P4wnP1...
https://github.com/mame82/P4wnP1 
... only to make it easy for folks to get started.

Anyway, DuckyScript is a mess. Gard to send raw keys, even harder to send non ASCII chars.
Building functions or branching is impossible, but needed for P4wnP1.

At least I added the ability to accept raw HID keycodes + modifiers to DuckEncoder, but the 8 month old PR was never reviewed 
https://github.com/hak5darren/USB-Rubber-Ducky/pull/54

Hak5 moved over to DuckToolkit on BashBunny, which is more robust but doesn't introduce missing features.

So as soon as I have the time to do it, I'll drop DuckyScript support and introduce something more usable.
This is especially needed to combine everything with MouseScript for P4wnP1.

I wouldn't advice anyone who starts a new HID keyboard project to use DuckyScript. It isn't flexible, quit but restricted and not robust.

mame82, 2017-10-09:
One more addition:
A boot protocol capable HID keyboard desriptor accepts 8 modifier bits and **6 keyboard bytes**. A RubberDucky only sends 1 modifier byte and **1 keycode byte** per HID input report. This restriction is represented in DuckyScript and makes it impossible to send multiple keycodes with the same modifier in a single report. This is of course a corner case, but playing multiple notes at the same time with a keyboard driven software synthesizer is impossible with DuckyScript ;-)

ChandraOrbit, 2017-10-10:
I've tried using https://exploitagency.github.io/Duckuino/index.html
For ordinary users it's a very helpful thing. Although the end result we have to change / adjust delay.
But I prefer writing myself because the Esploit script is more concise and broad.

additional
Some cases in my tests are using several machines with different OS,
1. Esploit has decreased speed in the payload delivery when the target computer is long enough to be used. And will be very responsive when the new Target computer is turned on.
2. Also if Esploit is long enough attached to the target then the payload shipping speed will be slightly reduced. Most likely the heat changes on the Esploit device.

So in making one charge, I have to do some tests with the above conditions.
So making one payload takes a long time (this is a constraint in my opinion)

exploitagency, 2017-10-10:
Focus more on text based commands in the payload and depend less on making your payload download an app, this is a keystroke injection attack after all.  I feel like your third response is referring to using the FTP server to download files to the victim PC, also if you must depend on some app to do your dirty work as I told you previously download it to the target using a HTTP transfer not FTP, you can find the link of any uploaded file from the exfiltrated files page even if you intentionally uploaded it via FTP and it didnt come from a victim pc.  I assume its likely an EXE file lets say test.exe the download link would be http://192.168.1.1/test.exe .  You get the idea.  I have not tested it but I assume HTTP to be faster and more reliable.  Don't even turn on FTP if you can help it. Use the HTTP exfiltration methods as well.  You have got to be creative with payloads. The hardware is somewhat limited.

ChandraOrbit, 2017-10-10:
Yes, For now Esploit will be maximized if more focus on text based command in payload because seen in terms of hardware minimalist.

I did some tests because I was a teacher aide, so I had to be able to summarize the questions from my students with all their strengths and limitations.

For Point 1 and 2 can be your consideration. It was the result of my experiments several times with OS and Machine with different specifications.
Maybe others will find different results.

exploitagency, 2017-10-10:
I hope it was a great lesson!  I wish I had such a device being demonstrated by my teacher while I was in school.  Keep up the good work.