exploitagency/ESPloitV2 #7: Appearance

harshil2105, 2017-09-18:
Make some new tabs,colours and style to your page

harshil2105, 2017-09-18:
It can't behave as a twin duck right??

ChandraOrbit, 2017-09-28:
Could it be incorporated with mmc so it behaves like USB / UFD, with the intent of camouflage also storage of payload or minimal data.

whid-injector, 2017-09-28:
At the moment the option of adding a mmc is not in consideration.
It will encrease price and volume. Which will lead in looking for another USB case.
That will encrease the price even more.
Overall the ratio benefit/time wasted is not worth it.

WHID was mainly created to wraponize usb gadgets or deployed by the attacker itself. 
The twin ducky style was not considered intentionally.

P.s. the original repo for Hardware related questions (i.e. Cactus WHID) is https://github.com/whid-injector/WHID

exploitagency, 2017-09-28:
Also right now in software if the victim and device are connected to the same network there are several WiFi based exfiltration options HTTP and FTP and a login credential phisher, see the ESPloit readme.

ChandraOrbit, 2017-09-28:
Is it possible to automatically connect to victim's network? I tried but could not and had to configure it manually with condition of knowing the condition of password, IP and SSID.

exploitagency, 2017-09-28:
You need to be on the same network for those exfiltration methods to work, meaning you need access to their WiFi credentials, so crack it before hand or much easier just do the reverse and have the victims pc connect to the esploit device's network when its running as an access point, see the exfiltration page when connected to the esploit web ui, I believe I show some examples of how to do this.  You can also exfiltrate to an online alternative like pastebin instead of straight to the device running esploit.

exploitagency, 2017-09-28:
This issue has gone way off course but here is an example of commands to force the victim to connect to ESPloit's network when in access point mode.

Windows: netsh wlan set hostednetwork mode=allow ssid="SSID-HERE" key="WIFI-PASSWORD-HERE"
Linux: nmcli dev wifi connect SSID-HERE password WIFI-PASSWORD-HERE

blackout314, 2018-05-15:
i'm a frontender :) if want i can do cosmetic improvements