exploitagency/ESPloitV2 #8: Exfiltrating Data block

ChandraOrbit, 2017-09-30:
Exfiltrating Data

I tried this feature but there are constraints that is when the router is used by the target attached firewall (in my trial using mikrotik router)
EsploitV2.5.3 can not upload files to list exfiltrade data using FTP

is there a solution or a way that can solve the problem?

exploitagency, 2017-09-30:
Two ways, think outside of the box.

Run ESPloit in access point mode, force the victim to connect to our access point, then you obviously bypass their routers firewall(although not one locally on their pc), exfiltrate.

Other option use HTTP exfiltration method.

So FTP works fine for you otherwise right?  If you dont use the proper modified FTP library as in instructions then it will not work in access point mode.

exploitagency, 2017-09-30:
One more note.  It supports PASSIVE mode FTP only! See example payloads for something verified to work.

ChandraOrbit, 2017-09-30:
I just realized that this is PASSIVE mode FTP only, thanks for enlightenment

exploitagency, 2017-09-30:
See the demo payloads for some ideas.  I will update documentation to reflect this when I get a chance.

exploitagency, 2017-10-01:
Updated docs

closed 2017-10-01 by exploitagency