exploitagency/ESPloitV2 #9: REQ: explanation for newbie

ChandraOrbit, 2017-10-01:
I am a teacher assistant at a high school, planning to buy some Cactus-WHID-WiFi-HID-Injector for students at my school.

On the wiki explained about:
1. Reset to default configuration / Restore device.
- Is there another way like a simpler script like using Powershell or CMD or a small tool to do it, because not always if there is a problem, we bring an Arduino laptop already installed IDE?

2. Credential Director ESPortal
Mentioned on the wiki about "Custom html templates can be uploaded for ESPortal sign-in credentials via FTP credentials."

For example in this case as Newbies who have purchased Cactus-WHID-WiFi-HID-Injector which since July 2017 all Cactus WHIDs are delivered with ESPloitV2 pre-loaded and ready for Plug-n-Hack,

- Can you provide an example of its application or its explanation on WIKI about the purpose "Custom html templates can be uploaded for ESPortal login via Incoming FTP credentials." (Note: in this case as a regular user or NEWBIES EDITION)?

I apologize for my ignorance, so many very basic questions or requests.

exploitagency, 2017-10-01:
1)
Arduino IDE is not needed to reset them, just connect to them via serial with baud set at 38400 I think it is or whatever it says in the readme.  Then issue the command "ResetDefaultConfig:"

2)
There are example templates here

https://github.com/exploitagency/ESPloitV2/tree/master/ESPortal-Templates

ESPortal is a phishing tool for credential harvesting

Set access point mode with a name like free wifi, turn on ESPortal, it spoofs pages with login templates, user tries to login, the username and password is saved to a file that can be viewed in the web ui.  Note: when running ESPortal the web ui no longer shows up on HTTP://IP but instead HTTP://IP/esploit

exploitagency, 2017-10-01:
Or do you mean what are the default credentials to log in?

WiFi
SSID: Exploit
PASS: DotAgency

Settings
User: admin
Pass: hacktheplanet

FTP:
User: ftp-admin
Pass: hacktheplanet

ChandraOrbit, 2017-10-01:
Forgive me for my unfamiliarity with the explanation in the Wiki, because my english is very ugly.

1. OK I understand, I will try again with some cases that I simulate my place.
 
2. Previously I thought that ESPortal is not permanent and can be changed by anyone with only upload via FTP to ESPloitV2 like uploading Payload.

Thanks.

exploitagency, 2017-10-01:
Your still not clear on point #2.

ESPortal can be turned on or off in the settings.

All pages on ESPortal have a very simple default page.  They appear as simple login prompts in several colors.

When you upload a properly named html file you can over ride these default templates with your own custom template.  See the readme for the naming convention.

Also it is very important that you use the provided templates as a basis for your template and retain the code required in the html template to make it function as intended.  As simple as the templates are it should be pretty obvious.

ChandraOrbit, 2017-10-01:
Ok, steady,
I had to hit my head to better understand it.
thank you for the time and explanation, On another occasion I might ask with all my stupidity. Thanks.

exploitagency, 2017-10-01:
Ask any questions that you need to succeed.

closed 2017-10-01 by exploitagency