exploitagency/ESPloitV2 #14: printing different key

M0tHs3C, 2017-11-08:
sorry, not at all sure this is an issue or a bad mistake that i have made, but with every payload on every pc the output that the ESPloit print is gnome'terminal when i typed in :gnome-terminal, so it does for all the other key like ù,è,:,_,é

exploitagency, 2017-11-08:
Change the language setting to match your locale.

It is set up by default for English in America.

You will need to modify the keyboard library and manually recompile and flash the 32u4 sketch.

closed 2017-11-08 by exploitagency

exploitagency, 2017-11-08:
Refer to the README

exploitagency, 2017-11-09:
> thread [2].
> Links:

What are you looking for?  Just click the help button from menu or read 
the readme in github repo.

Changing the Keyboard Locale

This is an easy to use solution from BlueArduino20 that is based off the 
work from NURRL at https://github.com/Nurrl/LocaleKeyboard.js
Linux: https://github.com/BlueArduino20/LocaleKeyboard.SH
Windows: https://github.com/BlueArduino20/LocaleKeyboard.BAT

-