exploitagency/ESPloitV2 #17: how would i dual boot whid cactus compatible firmware on a whid cactus ???

777anon666, 2018-04-07:
how would i dual boot whid cactus compatible firmware on a whid cactus ???

exploitagency, 2018-04-07:
I write the ESPloit software, you need to ask this in the hardware repo at whid.ninja

I have not seen this done and am not sure there is enough storage to create such a beast.

I will not attempt to support such a feature but am not against a custom bootloader type deal.

closed 2018-04-07 by exploitagency

whid-injector, 2018-04-07:
Hold on... why in this world you would like to have an arduino-based-dual-boot device?!
It is potentially possible, but too expensive in terms of ROI. 
Push requests are welcome though :smiley:

exploitagency, 2018-04-07:
Yeah... my thoughts.

exploitagency, 2018-04-07:
Cool idea though

whid-injector, 2018-04-07:
Indeed cool idea!
Like removing the atmega 32u4 and use only the ESP (with an ad-hoc USB stack). 
Doable, but the man hours used to implement it would kill the ROI of the entire idea behind WHID and ESPloit: a cheap HID offensive device available for anyone at an affordable price. :)

If I was still a teenager without job, kids and plenty of spare time.... I might have considered it...
But you know better than me how time is precious at our age :)

exploitagency, 2018-04-07:
For me my choices are sleep or code... I totally get the time issue.

777anon666, 2018-04-11:
It would be cool because say i want to keep esploit for some things but i like to use wifi ducky alot because of simple gui and easy scripting because you can use default usb rubber ducky code to build a script but esploit seems more stable (WiFi ducky loads fine on to the WiFi cactus but when i tell it to use a script or run a command it dose nothing is there a way you can help me solve this issue) Also is there a Firmware better then esportal that can support php based phisher for the wifi cactus WHID or is there a better website hosting with captive portal and credential harvester firmware for it thanks i know this is a lot sorry

777anon666, 2018-04-11:
I fixed the wifi ducky issue i was having but please answer the other parts of my above comment thank you sorry to bug you guys

exploitagency, 2018-04-11:
Use duckuino mode on esploit problem solved

exploitagency, 2018-04-11:
Also upload custom templates for esportal but no php

exploitagency, 2018-04-11:
You could write a payload to change the hosts file and then host your own phishing page.  You have to be creative or use more powerful hardware.  The point of this hardware is that it has enough features to perform many attacks but also is cheap enough to afford to lose a unit.

exploitagency, 2018-04-11:
To clarify ESPloit supports Ducky Script in DUCKUINO mode...

777anon666, 2018-04-11:
You are awesome thanks for taking the time with me to learn iam new to the whid world lol but a quick learner ...... Also could you please help with this if not question 1 at least question 2

1. Ok to start i love WiFi Ducky works great but will not save settings when i go to the settings tab and change Enable Script Execution on Startup: to enable by clicking the check mark box then click save it dont save the setting it will not even save the Auto-Script Name: in the box so what i had to do is manually edit the Arduino WiFi Ducky ESP.bin to have the settings i want to be enabled saved as enabled which works but when i check the setting tab after flashing that custom bin it dont show my changes but my changes are working even tho it says they are not enabled

2. Is there a way to add FTP Login to WiFi Ducky like Esploit has

exploitagency, 2018-04-12:
This is the wrong place to ask that question as I have nothing to do with that project but I hope you enjoy using ESPloit.

777anon666, 2018-04-12:
Thanks and i appreciate you for developing esploit

exploitagency, 2018-04-12:
Also if you want to run duckyscript at startup on esploit go to duckuino mode convert it to esploit code then save and upload that payload. Go to settings and enable it to run at startup

777anon666, 2018-04-12:
How did you add your ftp server code is there a github page you learned from? if you dont mind me asking

exploitagency, 2018-04-12:
I use a particular library check the readme it links it.