exploitagency/ESPloitV2 #18: #nicetoknow

hanshaze, 2018-04-27:
this is not really a problem, i'm just wondering if it would be possible for my mobile phone 
to execute code via the cactus wlan in, say, OTA on e.g.  a windows pc! 
So dongle is somewhere in the next room connected to a wall socket, laptop &mobile phone both in the same network where cactus is gateway.... 
SO AT  LEAST IS MY QUESTION: has the cactus plugged directly to the target to be interesting?
or is there some other way to handle that? who can enlighten me?

exploitagency, 2018-04-27:
The device must be physically connected to a target's USB port to deploy a payload.  Of course you can use the ESPortal mode (with the proper configuration) in this manner and perform a phishing exercise when both are on the same network.

closed 2018-04-27 by exploitagency

hanshaze, 2018-04-27:
Thanks....and can you go a little bit further into Detail please? My plan is to drop a meterpreter shell on the target without getting in contact.....so, how would you do that with phishing? Hta code in the portal html maybe?

exploitagency, 2018-04-27:
You obviously have to email a link to the phishing page or somehow communicate wih the victim that they need to visit the link.  Maybe upload an infected file to the esploit via ftp if it is a small enough file to fit in the limited storage else host it online. Create the page with a download link, name it properly and upload via ftp.  Then have the victim download the file and run it.

exploitagency, 2018-04-27:
Maybe even try using the network messenger service to send an alert to the victim.

hanshaze, 2018-04-28:
top!  mucias gracias ^^