exploitagency/ESPloitV2 #20: Functional devices, the possibility of wi-fi adapter

winner13, 2018-05-03:
Kind time of the day, I do not know if there is any support for this project, but I'll try to ask.

Can I use the device as a Wi-Fi adapter? Like entering data when connecting to another access point, the facility is working fine, but for full control I can not transfer the file, or use the network of the device itself (Exploit: DotAgency). Help to solve the problem with the control, can I somehow configure the WHID injector as well as the Wi-Fi adapter on the remote computer?

winner13, 2018-05-03:
I will clarify, the remote computer does not have a network, from peripheral devices, only a temperature sensor. With the help of the WHID injector, I can control the work, but I can not get data from the device.

P.S. Thought I was buying a wi-fi adapter, with keyboard emulator functions. It seems I was wrong.

exploitagency, 2018-05-03:
Your English is not so good(don't worry mine isn't either),
so I am having some trouble understanding your goal.

You are attempting to exfiltrate data?

If so just use the serial exfiltration method, no networking is needed with this method.

The device is not a "wifi adapter" in the sense I feel you are trying to use it.

If the computer has a wifi adapter then write a payload to connect it to the access point provided by ESPloit.  Or connect ESPloit to the same network the computer is on.

It works in one of two ways.

ESPloit provides an access point to allow access to the "ESPloit network" and obviously nothing external is available.

or

ESPloit is connected to an existing network and can be accessed from within that network by any device.

In neither configuration does ESPloit act as a "wifi adapter" for the computer who's USB port it is attached to.

I think that answers your question.  So I will close this issue.

closed 2018-05-03 by exploitagency