exploitagency/ESPloitV2 #21: Function of transferring a file of the ESploitV2.

winner13, 2018-05-03:
I apologize for my bad English.

I need to interact(get result.txt) with a remote machine that does not have a network. As it turned out, through the WHID injector, I can't receive data through the network itself Esploit, and it's sad.

Is it possible to use 3 megabytes of flash memory internally as a disk in Windows to transfer and receive files through it?

The algorithm is as follows:

1. Download the file getTEMP.cfg via ftp on Esploit.
2. Connect an internal flash memory(3 megabytes) as a Windows disk.
3. Use this playload:
`Rem:GET RESULT
Press:131+114
PrintLine:cmd
PrintLine:echo F:>>payload.bat
PrintLine:echo copy F:\getTEMP.cfg C:\area\cfg\>>payload.bat
PrintLine:echo start C:\area\control.exe>>payload.bat
PrintLine:payload.bat
PrintLine:echo copy C:\area\result.txt F:\>>result.bat
PrintLine:result.bat
PrintLine:exit`

Disk "F" - it's internal memory of Esploit, if there is a possibility to connect internal memory as a Windows disk.

4. And I already through wi-fi will be with ftp Esploit, broat the file with the results.
5. Set up the Ncrone, every hour, to perform the first 4 items, and take a file with the results.

exploitagency, 2018-05-03:
Mass storage is not a current feature.

Read the readme file and look at the example payloads. As I said previously try using the serial exfiltration method.  No networking is needed.

Everything is in the readme.

closed 2018-05-03 by exploitagency

exploitagency, 2018-05-03:
Wait, I misread the question.

I thought the victims computer could not access esploit as it had no networking capability. But now your saying you download a file via ftp(why not just download it via http).  Anyways this implies networking works and the computer does have networking capabilities.

If the computer has networking capabilities then use htttp exfiltration methods, ftp exfiltration methods or others.  Just look at the readme and example payloads.

winner13, 2018-05-03:
Is there a possibility, or hope, of the internal memory(2949250 bytes) of the WHID to connect to Windows as a flash disk??? 

I need to interact with the system from the outside, only through the WHID

exploitagency, 2018-05-03:
Not to sound offensive but I just can't understand you and am unsure if you have a solid grasp of what you have purchased.  Please READ THE README.  I put all of the information needed to be successful in the readme.

Does the victims computer possess the ability to connect to a WiFi network?

If yes then then the ESPloit can host its own private network by means of providing its own access point.  The computer must have a way to connect to this access point.  Thus look at my question above.

Did you read the readme?  Linked below is the relevant section of exfiltrating data.

https://github.com/exploitagency/ESPloitV2/blob/master/README.md#exfiltrating-data

Did you test the example payloads?  Look at the exfil win payloads.  Included is serial, http, ftp, you can even exfil data by blinking the led and analyzing the blinks and duration although that is not an included payload just a creative option.

https://github.com/exploitagency/ESPloitV2/tree/master/payloads

exploitagency, 2018-05-03:
Exfiltration is the process of retrieving the data.  Is this word where the confusion is coming into play?

exploitagency, 2018-05-03:
Also I suggest against using the FTP methods as HTTP methods are faster and only Passive mode ftp is supported.