exploitagency/ESPloitV2 #22: Blue LED blinks rapidly

hanshaze, 2018-05-07:
hi......im back with another question ;)
what does it mean, if theres no Exploit WLAN after a fresh reflash.......
instead the blue led just goes on and off instantly......

hanshaze, 2018-05-09:
Corey?

exploitagency, 2018-05-09:
The blue LED blinks fast during the flashing process.

On a fresh unit there is a roughly 90 seconds wait upon power on where the SPIFFS file system is formatted.

Then the rest of the program happens, load config, attack, loop, etc...

Never try to test the ESP before flashing the proper sketch to the Atmega32u4 as the 32u4 sketch turns on an enable pin to power up the ESP.  This is a common mistake people make, asking questions about the ESP not working when the ESP is not powered on.

Open a serial connection with the device, I believe I added feedback when you send commands.

What part of the process are you stuck on this time?

exploitagency, 2018-05-13:
Solved?  If so please close.

hanshaze, 2018-05-14:
Not solved.....

I've tried wifiducky and when it did not work (compiler constantly reports errors) I wanted to switch back to esploit .... I first flashed the programmer and then tried to prepare with esptool erase_flash, was not ........ something with timeout and header ...... So I han with the windows-tool on 0x00000 the 0x00000.bin and 0x010000 the 0x010000.bin flashed and then the ofw on 0x00000 .......

so far so good......

if I try now to recompile the esploit.bin with German keyboard I constantly get errors ....... screenshot is coming!

exploitagency, 2018-05-14:
You have modified the code, I do not support these situations.

Esplora.h?
Keyboard.h?
Hid.h?

Of course there are errors.

Why don't you try to compile the actual source?

The ESP is not a "keyboard", the 32u4 is the keyboard portion.  This is why it does not compile.

hanshaze, 2018-05-14:
ok.....
i have downloaded a new esploitv2 to my desktop.....
then i reflashed this bin file here --->

[image]

next step was to open arduino ide to customize the ESP_SKETCH, right?
i did so and here are my used settings --->

[image]

after i clicked on "compile/export" it gave me following error --->

[image]

hope you can help, because i have no idea, i reinstalled the ide as well and i dont know what i can do ^^

exploitagency, 2018-05-14:
How did you customize the ESP sketch and why?

The keyboard portion is the 32u4 sketch.  You need to customize the keyboard library globally or redefine the include and copy the keyboard library to the 32u4 sketch folder and modify it there.

exploitagency, 2018-05-14:
You should just flash the precompiled ESP binary from here.
http://esploitv2.exploit.agency/releases

hanshaze, 2018-05-14:
....and again, thanks alot mate ^^

[image]

closed 2018-05-14 by hanshaze