exploitagency/ESPloitV2 #26: failed to request a device descriptor

moneak, 2018-08-09:
After I followed these steps 

> Load the esp8266Programmer sketch from the flashing folder.
Select Tools - Board - "LilyPad Arduino USB".
Select the Port your device is connected to under Tools - Port.
Upload the sketch.

from your instructions, my ESP stopped responding, and I'm getting now just that  "failed to request a device descriptor". Windows does not see the device

ChandraOrbit, 2018-10-07:
I have experienced this several times, the actions I have done are as follows
1. Make sure the Driver is installed if you use Windows and the WHID has been detected, please see Device Manager => Port
2. Reset the WHID with a small Magnet, I get it from my friend's workshop.
- plug WHID to your computer
- hold the magnet and close the magnet to your WHID then shake right or left quickly
- Automatically reset the WHID to ZERO
- do Flash from the beginning

I hope you understand my information, sorry my English is very bad.

demian123, 2019-04-18:
I'm having the exact same problem with my Cactus WHID, it's not being recognized by any device. Already tried with three different magnets and nothing seems to happen, what am i doing wrong?