exploitagency/ESPloitV2 #32: Not displaying AP?

j-dizzle1, 2019-01-27:
I've reflashed the USB stick twice now with no solution.  When I first got it I set it to be a hidden AP and now the light just flashes on and off, and I can't connect to it, I tried resetting the default config, and it says it is relaying the command but then nothing happens.  I have flashed it twice to no success.  Please help Thanks

exploitagency, 2019-01-27:
Lets back track and you describe the steps you took to "flash it" and then lets figure out where things went wrong.

j-dizzle1, 2019-01-28:
I used the Initial flashing, however when that didn't work I used the [whid injector repo ](https://github.com/whid-injector/WHID/wiki) and uploaded the programmer sketch and then erased all the data.  I will attempt to reflash now, I start by loading the esp8266Programmer sketch to the arduino and upload that to the lilypad usb on port /dev/cu.usbmodemHIDFG1 (am using mac), now that that is over the port becomes /dev/cu.usbmodem14101.  I open up the terminal (I have already installed all the needed git stuff but I will erase it and start over) I open up the espcode.ino and set the board to generic esp8266 and the flash size to 4M (3M SPIFFS) then I compile the binary.  Then I copy the binary and paste it in to the folder contain esptool.py and from the command line run python esptool.py --port=/dev/cu.usbmodem14101 --baud 115000 write_flash 0x00000 ESP_Code.ino.generic.bin --flash_size 32m.  That is successful so on the same port I change the board to lilypad usb and upload 32_u_4 code. Done uploading and the blue light is flashing, but no AP, I also tried connecting a hidden network to no avail :( and I doing something wrong?  Thank you so much.  You are the best