exploitagency/ESPloitV2 #34: Migrate to platformio

volca, 2019-01-29:
Hi,

I've recently migrated the latest ESPloitV2 code to platformio.  So we can flash the WHID board with one line.

```
./flash.sh
``` 

or 

```
flash.bat
```

Please check the code [here](https://github.com/volca/ESPloitV2/tree/master/source)

exploitagency, 2019-01-29:
Be sure that your pulling the proper versions of all the libraries. It actually relies on older libraries than the current versions.

volca, 2019-01-29:
The libraries are up to date. The ESP8266 Arduino core is v2.4.2

```
pio lib list

ArduinoJson
===========
#ID: 64
An elegant and efficient JSON library for embedded systems

Version: 5.13.4
Homepage: https://arduinojson.org/?utm_source=meta&utm_medium=library.json
Keywords: json, rest, http, web
Compatible frameworks: arduino
Compatible platforms: atmelavr, atmelsam, espressif32, espressif8266, intel_arc32, microchippic32, nordicnrf51, nordicnrf52, ststm32, teensy, timsp430
Authors: Benoit Blanchon

ESP8266FtpServer
================
#ID: 992
Very Simple FTP server for SPIFFS on esp8266

Version: 1.0.1
Keywords: communication
Compatible frameworks: arduino
Compatible platforms: espressif8266, espressif32
```

exploitagency, 2019-01-29:
Initial Flashing
Download and Install the Arduino IDE from http://www.Arduino.cc
Open Arduino IDE.
Go to File - Preferences. Locate the field "Additional Board Manager URLs:"
Add "http://arduino.esp8266.com/stable/package_esp8266com_index.json" without quotes.
Click "Ok"
If Arduino IDE gives you the following error:
"Error downloading http://arduino.esp8266.com/stable/package_esp8266com_index.json"
Use "https://github.com/esp8266/Arduino/releases/download/2.3.0/package_esp8266com_index.json" instead.
Select Tools - Board - Boards Manager. Search for "esp8266".
Install "esp8266 by ESP8266 community version 2.3.0". Click "Close".
Select Sketch - Include Library - Manage Libraries. Search for "Json".
Install "ArduinoJson by Benoit Blanchon version 5.11.0" and click "Close"   Download https://github.com/exploitagency/esp8266FTPServer/archive/feature/bbx10_speedup.zip
Click Sketch - Include Library - Add .ZIP Library and select bbx10_speedup.zip from your Downloads folder.
The Arduino IDE is now configured and ready for the code.

Use git to clone this repo: https://github.com/exploitagency/ESPloitV2.git
or
Download/extract the repo as a zip file: https://github.com/exploitagency/ESPloitV2/archive/master.zip

Load the esp8266Programmer sketch from the flashing folder.
Select Tools - Board - "LilyPad Arduino USB".
Select the Port your device is connected to under Tools - Port.
Upload the sketch.

Open the ESP_Code sketch from the source folder.

exploitagency, 2019-01-29:
esp8266 by ESP8266 community version 2.3.0

ArduinoJson by Benoit Blanchon version 5.11.0

Ftp server https://github.com/exploitagency/esp8266FTPServer/archive/feature/bbx10_speedup.zip

volca, 2019-01-30:
> esp8266 by ESP8266 community version 2.3.0

For platformio, I can change [the platformio.ini](https://github.com/volca/ESPloitV2/blob/master/source/esp/platformio.ini#L12) to force use version 2.3.0. Maybe we can also take a try for latest stable v2.4.2?

```
platform = espressif8266@1.5.0
```
I've changed the library dependency to the specify version

volca, 2019-03-04:
see https://github.com/volca/ESPloitV2/commit/44b0417dde354d8ebc2e1ae1e0d16e1b1bd2c1b8

closed 2019-03-04 by volca