exploitagency/ESPloitV2 #39: When I change the WiFi settings the cactus WHID stop working

k0stQs, 2019-02-18:
When I change the WiFi settings the cactus WHID stop working, to be precise the light just flashes on and off and I can't connect to it.

I followed the steps below to flash it:

---------------------------------------------------------------------------------------
Initial Flashing

Download and Install the Arduino IDE from http://www.Arduino.cc
Open Arduino IDE.
Go to File - Preferences. Locate the field "Additional Board Manager URLs:"
Add "http://arduino.esp8266.com/stable/package_esp8266com_index.json" without quotes.
Click "Ok"
If Arduino IDE gives you the following error:
"Error downloading http://arduino.esp8266.com/stable/package_esp8266com_index.json"
Use "https://github.com/esp8266/Arduino/releases/download/2.3.0/package_esp8266com_index.json" instead.
Select Tools - Board - Boards Manager. Search for "esp8266".
Install "esp8266 by ESP8266 community version 2.3.0". Click "Close".
Select Sketch - Include Library - Manage Libraries. Search for "Json".
Install "ArduinoJson by Benoit Blanchon version 5.11.0" and click "Close"   Download https://github.com/exploitagency/esp8266FTPServer/archive/feature/bbx10_speedup.zip
Click Sketch - Include Library - Add .ZIP Library and select bbx10_speedup.zip from your Downloads folder.
The Arduino IDE is now configured and ready for the code.

Use git to clone this repo: https://github.com/exploitagency/ESPloitV2.git
or
Download/extract the repo as a zip file: https://github.com/exploitagency/ESPloitV2/archive/master.zip

Load the esp8266Programmer sketch from the flashing folder.
Select Tools - Board - "LilyPad Arduino USB".
Select the Port your device is connected to under Tools - Port.
Upload the sketch.

Open the ESP_Code sketch from the source folder.
Select Tools - Board - "Generic ESP8266 Module".
Select Tools - Flash Size - "4M (3M SPIFFS)".
Select Sketch - "Export Compiled Binary".

Now flash the firmware to the ESP-12S chip using one of the following tools.
Linux: https://github.com/AprilBrother/esptool
Example: python esptool.py --port=/dev/ttyACM0 --baud 115000 write_flash 0x00000 ESP_Code.ino.generic.bin --flash_size 32m
Windows: https://github.com/nodemcu/nodemcu-flasher

NOTE: Do not try to connect to the access point or test anything yet, the device won't work until after the next step.

Finally open the Arduino_32u4_code sketch from the source folder.
Select Tools - Board - "LilyPad Arduino USB".
Select the Port your device is connected to under Tools - Port.
Upload the sketch.

Your ESPloit is now ready to configure and or use.

--------------------------------------------------------------------------------------------

Ι also use the precompiled binaries (https://github.com/exploitagency/ESPloitV2/releases), but without result.

None of the above steps did fix the problem. WHID continues to flashes on and off, and of course no wireless network (SSID "Exploit") appears. 

Can you please help me?

Thanks in advance

k0stQs, 2019-02-18:
I finally found the solution...

k0stQs, 2019-02-19:
I used the version 2.7.41 to flash flash it and then upgraded it to the release v2.7.51.

k0stQs, 2019-02-19:
I used the version 2.7.41 to flash it and then upgraded it to the release v2.7.51.

minkione, 2019-02-19:
Thanks, someone may find it useful. :)
Strange thought that you couldn't flash straight the v.2.7.51

k0stQs, 2019-02-19:
I think that the problem that I couldn't flash it straight to the v.2.7.51 is "esp8266 version 2.3.0" or "ArduinoJson version 5.11.0", But I'm not sure.

mcjeikk, 2021-07-06:
I followed this steps for flash my Cactus WHID but it doesn't work, the Wifi doesn't appear. I have 3 of them and all the USBs is the same.

HankDaHacker, 2022-04-17:
There isn't even and flashing lights on mine. Did I recieve a faulty dud?