exploitagency/ESPloitV2 #42: Worked until I changed the SSID and password. The the SSID became invisible and blue light started blinking

vcchm, 2019-03-08:
The device came pre-installed. It worked correctly until I changed the SSID and password. The the SSID became invisible and blue light started blinking and I could not use it anymore.

I tried reflashing it using the procedure proposed here : https://github.com/exploitagency/ESPloitV2
I had no error message installing the loader, the ESP flashed correctly as well as the third code (no error message) but the device is still not working
.
I tried erasing the internal memory by flashing a 4MB blank file before flashing the ESP code. It does not work either. I tried installing a release binary (ie not compiling the ESP code myself) and it does not work either.

What did I do wrong ? Is there a bug in the code when you change the SSID name ?

vcchm, 2019-03-08:
Ok, I flashed it using NodeMcuFirmware Flasher and 57600 bauds speed, and it worked.
IMHA the only things that need to be fixed is
    1 - mentioning the limited speed in the flashing manual, and maybe the need the clear the flahs when re-installing
    2- fixing the bug that bricks the WHID when changing the settings. What I have done is changing the arduino code to choose a different default name.

hexxter, 2019-03-10:
jup the same problem. After saving the settings is the wlan down and the esp start blinking.

unbaiat, 2019-03-14:
@vcchm can you please post the fix? tried your suggestion and I'm still stuck with the stupid blue light every 10 sec