exploitagency/ESPloitV2 #43: help, does not work

OverLS, 2019-04-09:
What to do when rev3 rev3. Maybe there is a manual?
After performing actions from the existing - the LED is on and that's it.
There are no fawns. SSID does not appear.

===============================================

Как быть когда ревизия rev3. Возможно есть manual?
После выполнений действий с существующего - горит светодиод и все.
Мирцаний нет. SSID не появляется.

OverLS, 2019-04-09:
[image]
[image]

exploitagency, 2019-04-11:
Read the manual the ssid is hidden.

closed 2019-04-11 by exploitagency

OverLS, 2019-04-11:
> Read the manual the ssid is hidden.
> 
> https://github.com/exploitagency/ESPloitV2/blob/master/README.md#esploitv2

=======================================================

I read everything, according to the [instructions](https://github.com/exploitagency/ESPloitV2/blob/master/README.md#initial-flashing) I made the firmware. There were no errors during the firmware process. now the LED is constantly shining

exploitagency, 2019-04-12:
Im assuming you did not follow instructions and use correct library versions?

Use the compiled firmware from the release page