exploitagency/ESPloitV2 #56: Layout problem

Ezo666, 2019-12-08:
How to solve keyboard layout problem?
In my region there is Russian layout is default, but some times user can switch layout to En, well you can not know what layout on the system at the moment. But if layout is Ru, then payload works incorrect, because all commands are with Russian symbols.
I tryed to use Alt+NUMPAD ASCII, but I can't find correct codes for numpad digits(usual digits not work)

irraz, 2020-04-20:
Hello,
I have the same question, I need change keyboard layout to whid-injector for input mode in ESPloit
Thanks,

irraz, 2020-04-21:
Solution is documented: https://github.com/whid-injector/WHID/wiki/Keyboards-Layout in FRENCH Layout section